MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #153: 请求参数错误10000，但是安装设置始终提示安装包损坏(错误码-2)，有遇到过的么

lmb1113, 2024-07-29:
- 机型k50pro
- 开发版1.0.24.7.25.DEV
- 请求参数错误10000
- 恢复出厂试过，无效
- 关闭miui优化，从文件管理安装，无效
- 根据教程安装settings.apk始终提示安装包损坏(错误码-2)
- 全网的安装包都试过了，都是无法安装
- ADB安装也尝试过，无法安装
- 删除过设置，但是也无法安装上

[image]

[image]
[image]

xjz1994, 2024-08-14:
> * 机型k50pro
> 
>     * 开发版1.0.24.7.25.DEV
> 
>     * 请求参数错误10000
> 
>     * 恢复出厂试过，无效
> 
>     * 关闭miui优化，从文件管理安装，无效
> 
>     * 根据教程安装settings.apk始终提示安装包损坏(错误码-2)
> 
>     * 全网的安装包都试过了，都是无法安装
> 
>     * ADB安装也尝试过，无法安装
> 
>     * 删除过设置，但是也无法安装上
> 
> 
> ![Screenshot_2024-07-29-23-55-12-400_com miui packageinstaller](https://private-user-images.githubusercontent.com/39643887/353142474-7d8ca342-4870-421b-80bc-d793c6cefe09.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM2MjEyNjQsIm5iZiI6MTcyMzYyMDk2NCwicGF0aCI6Ii8zOTY0Mzg4Ny8zNTMxNDI0NzQtN2Q4Y2EzNDItNDg3MC00MjFiLTgwYmMtZDc5M2M2Y2VmZTA5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE0VDA3MzYwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzNTIyNjg2ZWVlNmNjODFjOGNhN2IyM2FiZmM0NmU1NjAzZTg5ZTEzNzY4ZGQ2YWQyOGU5OTM2NjE3OGNjMGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5mkWmOxopcjBw1kmOF14CMjmUsV2DOlxnilwqQgK6kE)
> 
> ![5a740eeca1e92b9072b675ae96b50ee4](https://private-user-images.githubusercontent.com/39643887/353140926-f4cf3140-f1ec-4e58-8133-b7960e5943ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM2MjEyNjQsIm5iZiI6MTcyMzYyMDk2NCwicGF0aCI6Ii8zOTY0Mzg4Ny8zNTMxNDA5MjYtZjRjZjMxNDAtZjFlYy00ZTU4LTgxMzMtYjc5NjBlNTk0M2VlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE0VDA3MzYwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3OTc3YzkxODM0MmU1MjIzNzc3OTAwMDVkMjBiYzU1YmZhYTM3YmU1ZTE2YjQ5ODcxZGE4NGExNjIyOWVlZTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RFn4wCjPcaH9wpGZbqBEhQfMhQYJ-u8NzHz2sJUdKbY) ![image](https://private-user-images.githubusercontent.com/39643887/353141523-edbcd79c-fbad-461a-8b54-71c93b19ee62.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM2MjEyNjQsIm5iZiI6MTcyMzYyMDk2NCwicGF0aCI6Ii8zOTY0Mzg4Ny8zNTMxNDE1MjMtZWRiY2Q3OWMtZmJhZC00NjFhLThiNTQtNzFjOTNiMTllZTYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE0VDA3MzYwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlYTMwNDZjMGYxYjMwZDkwNTU3ZWExNDFkYzMyMzc3N2JlYTdmMjViNjMzM2MzMzYwZGE3ODBkNjYyNDIzNzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.keAEMddi6bFLBmvj62Z14IS5w4K1_BwqeWaPhUcYTwQ)

大佬，想问下你这些setting安装包是在哪里下载的

xjz1994, 2024-08-14:
#153 大佬想问下你的setting安装包是在哪里下载的

lmb1113, 2024-08-14:
> #153大佬想问下你的设置安装包在哪里下载的

https://pan.quark.cn/s/16bbc6281511

停更:
https://www.123pan.com/s/625SVv-BYxbh.html

xjz1994, 2024-08-14:
> > #153大佬想问下你的设置安装包在哪里下载的
> 
> https://pan.quark.cn/s/16bbc6281511
> 
> 停更: https://www.123pan.com/s/625SVv-BYxbh.html

感谢大佬，我用作者在release中提供的安装包，直接用adb安装就成功了，没有关闭系统优化。 
机型13 ultra， OS: 1.0.12.0

lmb1113, 2024-08-14:
最新版本秒BL工具，听说是开发版最新系统可以直接秒。工具链接
https://pan.quark.cn/s/a2d853b08dc0

据说可以秒，但是已弃坑，没试

closed 2024-08-14 by lmb1113