MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #159: Mobile Data Disconnecting midway through (Redmi Note 12 4G)

Andy1207-DEV, 2024-08-09:
So, im not sure if this is something that is common or it's just me or my mobile data disconnects right when i click the bind account button, then the script just stops at "Sending POST request" and gives a network error or "Parameter error: 10000".
[image]

BlazingRiser, 2024-08-19:
Do not use same internet connection(hotspot or usb thetering) from the device you want to use the script for. It disconnects from internet while blocking binding request. Script should have a stable connection during the whole process. Use another device for hotspot in pc.

bratok0000, 2024-08-23:
Hello! I have the same problem. I have a separate Internet and again this error and disconnection of the Internet. Help what to do?

BlazingRiser, 2024-08-23:
Here refer to this settings apk for request parameter error https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/82#issuecomment-2113933639. As for check connection, you can set global flag to true in bypass.php and see if that helps.