MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #160: The CMD return  a log "Request parameter error (10000)"

Kkongry0819, 2024-08-09:
I installed the latest release version of PHP (php -8.3.10)

My phone, which I wanted to unlock the BL, is a Xiaomi 12

I believe that I have configured ‘php.ini’ correctly

How can I fix this problem?
[image]

FeelLiao, 2024-08-12:
same problem here

ahadobas, 2024-08-12:
> update settings.apk first.

How?
Or where can I download from?

ahadobas, 2024-08-12:
> update settings.apk first.

Downloaded, tried to install, but got: 
[image]

ahadobas, 2024-08-12:
> See this reason why error 10000. Use google translate. [#144 (comment)](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/144#issuecomment-2251043736)

As I seen:
---------
This is because Xiaomi changed the log encryption algorithm from pure AES to RSA+AES hybrid encryption in the new version of the system. 
RSA asymmetric encryption cannot be decrypted without a private key, so the script cannot operate normally.
---------

Is there a way to get the "private key" from my phone and insert it into the script?

yuniers, 2024-09-05:
I'm in same situation.

```js
php-adb php bypass.php
[2024-09-04] [18:44:30] [INFO] - ************************************
[2024-09-04] [18:44:30] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-09-04] [18:44:30] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-09-04] [18:44:30] [INFO] - ************************************
[2024-09-04] [18:44:30] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-09-04] [18:44:30] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-09-04] [18:44:30] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-09-04] [18:44:30] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-09-04] [18:44:30] [INFO] - My Blog: https://www.neko.ink/
[2024-09-04] [18:44:30] [INFO] - ************************************
[2024-09-04] [18:44:30] [INFO] - Starting ADB server...
[2024-09-04] [18:44:30] [INFO] - Processing device lvcuamdahizhpfvs(device:ruby)...
[2024-09-04] [18:44:30] [INFO] - Finding BootLoader unlock bind request...
[2024-09-04] [18:44:30] [INFO] * Now you can bind account in the developer options.
[2024-09-04] [18:44:35] [INFO] - Account bind request found! Let's block it.
[2024-09-04] [18:44:35] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in /home/sosa/Downloads/php-adb/bypass.php on line 333
[2024-09-04] [18:44:35] [INFO] - Sending POST request...
[2024-09-04] [18:44:37] [INFO] - Request parameter error (10000)
```

matiasramirez1601, 2024-09-13:
I'm facing the same issue on my poco F5 with global rom on hyperOS 1.0.8.0.UMRMIXM

`[2024-09-13] [22:28:32] [INFO] * Now you can bind account in the developer options.
[2024-09-13] [22:28:44] [INFO] - Account bind request found! Let's block it.
[2024-09-13] [22:28:44] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333
[2024-09-13] [22:28:44] [INFO] - Sending POST request...
[2024-09-13] [22:28:45] [INFO] - Request parameter error (10000)`