MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #173: How to force downgrade settings

noxer04, 2024-09-15:
I've tried all kinds of settings apk (the chinese html doesnt have the one specific for garnet), installing them through package manager or adb install, still get "can't downgrade" or "com.android.settings is not updatable". I've also uninstalled settings and security apk through adb and tried to install an older settings but nothing, not updatable. 

So what to do? Are we stuck? My device is on hyperos 1.0.18.0UNREUXM, came with miui.

Is there a way to downgrade to miui with locked bootloader?

prgmr, 2024-09-26:
I have the same problem.
installed hyperos 1.0.12.0.UNRMIXM firmware
unable to update Settings.apk
When installing via file manager I get the error “App not installed as package appears to be invalid”.
When installing via adb, the error “Failure [INSTALL_FAILED_INVALID_APK: Package com.android.settings are not updateable.]” occurs.

FlyraXD, 2024-10-05:
Any updates on how to fix this?

prgmr, 2024-10-24:
I got approval through xiaomi community app.

MihailoDevelops, 2024-10-24:
@prgmr At what time did you send the request and have you activated Find My Device and SIM Sync? Could you share more details what exactly have you done?

dmrangel, 2024-10-24:
@MihailoDevelops you have to send the request 00:00 CST (China Standart Time)

MihailoDevelops, 2024-10-24:
@dmrangel I've been trying to do that for a few weeks now, but every time I would get limit reached or account error.

dmrangel, 2024-10-24:
Me too, ive tried 5 times and now i need to wait till 10/30

nixvsz, 2024-12-22:
Any solutions?

wanchaigame, 2024-12-28:
what else

lfox02, 2025-01-14:
Anybody who needs it, an uninstalled app can be brought back with (in this case)

adb shell pm install-existing com.android.settings

No restrictions, you're done in a second. You could even do it from within a terminal on the phone itself. Cheers

LaShids, 2025-02-27:
In theory, would it work if i uninstalled settings via adb and then installed it again?

lfox02, 2025-02-27:
@LaShids Well that's exactly what I did, so yes. The latest HyperOS will prevent you from installing a different one, but at least you can control stuff on the phone with the stock one.

LaShids, 2025-02-28:
@lfox02 what command did ya use? Every one i try says "Failure [DELETE_FAILED_INTERNAL_ERROR]"