MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #180: Poco F6 - Request parameter error (10000)

BgenkBlackisme, 2024-09-22:
Poco F6 - Request parameter error (10000)

[image]

BgenkBlackisme, 2024-09-22:
my hyperos
[image]

fillahfirdausyah, 2024-09-22:
same issue here

AmihayHassan, 2024-09-23:
same here, Poco F6 running HyperOS 1.0.7.0 UNPMIXM

added `var_dump($args);` in line 333

this is the output

[image]

BgenkBlackisme, 2024-09-23:
I will try tomorrow, thanks for your respond bro

BgenkBlackisme, 2024-09-24:
Still Cant Solved

AmihayHassan, 2024-09-24:
Thank for the quick reply @MlgmXyysd
Is there anything we can do to help?

AmihayHassan, 2024-09-24:
> Still Cant Solved

Thank for the quick reply, is there anything we can do to help?

BgenkBlackisme, 2024-09-24:
after quota limit, i got account error, wtf

BlazingRiser, 2024-09-27:
Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue. Although turbo 3 had issue with error 30003.

AmihayHassan, 2024-09-28:
> Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue. Although turbo 3 had issue with error 30003.

Not speaking Chinese but somehow managed to get there, anyway, I couldn't download it, and other versions (from apkmirror) are failing to install

JarJarSphynx, 2024-10-07:
> Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue. Although turbo 3 had issue with error 30003.

except that there is no apk in this list for the F6, only the turbo3.

AmihayHassan, 2024-10-08:
> > Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue. Although turbo 3 had issue with error 30003.
> 
> except that there is no apk in this list for the F6, only the turbo3. 

I think this is basically the same device..

betamaster8, 2024-10-19:
We would have a fix for Poco F6 5G soon for sure - kudos to the fantastic project
It is just a matter of time

marufmoinuddin, 2024-10-25:
> We would have a fix for Poco F6 5G soon for sure - kudos to the fantastic project It is just a matter of time

can we expect a fix for redmi note 13 pro 5g too?

JarJarSphynx, 2024-11-15:
> I think this is basically the same device..

No it's literally not. you can't install the turbo setting apk on the f6 in file manager, adb or anyway, besides if you try with a mi account on the f6 it will keep prompting you to try again after two weeks. so everyone including the dev who made this fix pretty much left the poco f series for dead here. glad I sold it before it was too late.

BlazingRiser, 2024-11-15:
> > > > Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue. Although turbo 3 had issue with error 30003.
> > > 
> > > 
> > > except that there is no apk in this list for the F6, only the turbo3.
> > 
> > 
> > I think this is basically the same device..
> 
> No it's literally not. you can't install the turbo 3 setting apk on the f6 in file manager, adb or any way, besides if you try with a mi account on the f6 it will keep prompting you to try again after two weeks regardless of whether your account region was set to global or your country so you will never be able to unlock it. tl;dr: everyone including the devs who made this fix pretty much left the f6 for dead here.

It's the same device, around August Xiaomi blocked the bypass for three devices shipping hyperos out of the box, before August the bypass used to work for f6, you can still unlock via official/or paid service. You can check devices which are blocked in [status.md](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/blob/master/status.md)

betamaster8, 2024-11-15:
Basically has moved of - to different ecosystem
No more "Xiaomi" in all my connection,, extended connections and professional connection

I have made people more than 400+ users, leave xiaomi devices who all once enjoyed xiaomi, as I configured their devices the way they wanted it to work.

Now a new ecosystem - No more xiaomi for all my connections

Loss for xiaomi - nothing for us to loose.
@xiaomi, @poco, @redmi and all their devices for making our lives difficult 

Thank you guys for this fantastic project!
Cheers
Bm

JarJarSphynx, 2024-11-16:
> It's the same device, around August Xiaomi blocked the bypass for three devices shipping hyperos out of the box, before August the bypass used to work for f6, you can still unlock via official/or [...]

Nope, same hardware maybe, not the same target firmware. I trie the turbo 3 apk myself, tried to install by adb and even edited the apk myself, when it's installed it force closes, it doesn't even have the same gui/icon btw. as for the f6 you can't unlock it by the official method for the issue I mentioned and not even paid services like ncunblock have a service for the f6.
tl;dr if anyone bought the f6 past august and expects to unlock it, sell it. don't bother trying to unlock it.

Mostwest, 2024-11-26:
Nice, I just bought my Poco F6 3 days ago...
I'm not even able to use the regular activation, this is what happens at 0:00 China time

The Xiaomi Community app can't even load the page because the servers are oversaturated by requests

I hoped to find a way but...

[image]

vpolinov, 2024-12-01:
> No more "Xiaomi"

support 100%

cute-omega, 2024-12-29:
here mentioned error code `10000` https://github.com/xbn2002/Xiaomi-HyperOS-BootLoader-Bypass/tree/patch-1
and for anyone who need to downgrade setting app, here may be something you need https://github.com/code4fun1024/setting-app-for-hyperos

JarJarSphynx, 2025-01-15:
> here mentioned error code `10000` https://github.com/xbn2002/Xiaomi-HyperOS-BootLoader-Bypass/tree/patch-1 and for anyone who need to downgrade setting app, here may be something you need https://github.com/code4fun1024/setting-app-for-hyperos
No it doesn't work. xiaomi blocked downgrading the setting app, there's literally no way you can install this. 

I've been saying this since 2024. 

[image]

and on top of that, they blocked redmi turbo 3 from unlocking without an account, and you can't unlock with an account no matter how long your account is since it's creation because the community app will ask you to try again after two weeks!

just never purchase anything from xiaomi again, genuine advice.

> I think this is basically the same device..

Mostwest, 2025-01-16:
> here mentioned error code `10000` https://github.com/xbn2002/Xiaomi-HyperOS-BootLoader-Bypass/tree/patch-1 and for anyone who need to downgrade setting app, here may be something you need https://github.com/code4fun1024/setting-app-for-hyperos

Yeah this stopped working around september for Poco F6/redmi Turbo updated devices.

Downgrading doesn't work. 
You need an account with permission to unlocked it.
I went for the paid way but the guy who always unlock phones for many wasn't able to do it for my phone 2 weeks ago, he did refund me.

JarJarSphynx, 2025-01-18:
> Yeah this stopped working around september for Poco F6/redmi Turbo updated devices.
>
> Downgrading doesn't work. You need an account with permission to unlocked it. I went for the paid way but the guy [...]

yeah well as I've already mentioned like 5 times before not even their official app works and it'll just ask you to try again after two weeks even if you apply on time (00:00 in mainland) with only sim data activated and global region

[image]