MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #183: Request parameter error (10000)

keenan316, 2024-09-26:
Hi,

I'm using a Poco X6 5G with latest HyperOS 1.0.16.0 UNREUXM, so I also can't downgrade settings.apk.

I created my Xiaomi account in 2015, but still it's almost impossible to just unlock the bootloader.

Is there any other way to get past this error other than jumping through Xiaomi's ridiculous hoops?

Thank you

Snarov, 2024-09-27:
The same crap is with my Note 13 pro 5G Global. Unlocking Xiaomi bootloader has never been so tricky.

Any help wll be appreciated

BlazingRiser, 2024-09-27:
Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue.

Tigercat000, 2024-09-28:
> Go in pull requests, there is one open post, inside there is link to 123pan, download your device specific settings apk from there that should fix this issue.

Same as X6 5G try all on 123pan setting.apk , is not working now, anyone have update news?

BootloaderJunkie, 2024-09-28:
Same here. After downgrading / replacing settings with 123pan variant or a variant from apkmirror, I get:
"The system is being upgraded. Please try again later. (30003)"

blackmatch, 2024-09-29:
For error code 10000, you can downgrade Setting using this file: https://drive.google.com/file/d/1T8XjD6mO3TkwMSX8ssEFOHUVn91tN8op/view?usp=sharing

and then try to unlock again.

Tigercat000, 2024-09-29:
> For error code 10000, you can downgrade Setting using this file: https://drive.google.com/file/d/1T8XjD6mO3TkwMSX8ssEFOHUVn91tN8op/view?usp=sharing
> 
> and then try to unlock again.

Thanks my friend , can you teach me how to make this apk? can make it myself?

keenan316, 2024-09-29:
> For error code 10000, you can downgrade Setting using this file: https://drive.google.com/file/d/1T8XjD6mO3TkwMSX8ssEFOHUVn91tN8op/view?usp=sharing
> 
> and then try to unlock again.

It doesn't work. When trying to install it's the same as all other setting.apk available..

"App not installed as package appears to be invalid"

Tigercat000, 2024-09-29:
> It doesn't work.

It doesn't work of me too..same result, can you modify ?

Tigercat000, 2024-09-29:
> For error code 10000, you can downgrade Setting using this file: https://drive.google.com/file/d/1T8XjD6mO3TkwMSX8ssEFOHUVn91tN8op/view?usp=sharing
> 
> and then try to unlock again.

It doesn't work of me too..same result "App not installed as package appears to be invalid", can you modify ?

blackmatch, 2024-09-30:
> App not installed as package appears to be invalid

I guess I used the wrong ROM to extract the apk file, can anyone tell me what the ROM are you use ? so that I can extract the correct apk file.

blackmatch, 2024-09-30:
> > For error code 10000, you can downgrade Setting using this file: https://drive.google.com/file/d/1T8XjD6mO3TkwMSX8ssEFOHUVn91tN8op/view?usp=sharing
> > and then try to unlock again.
> 
> Thanks my friend , can you teach me how to make this apk? can make it by myself?

It's very complex and will use many tools, but the main steps are:
* get the `payload.bin` file from the earliest HyperOS ROM.
* get the `system_ext.img` from `payload.bin`
* get the apk file from `system_ext.img`

Smartslabs, 2024-09-30:
> > > For error code 10000, you can downgrade Setting using this file: https://drive.google.com/file/d/1T8XjD6mO3TkwMSX8ssEFOHUVn91tN8op/view?usp=sharing
> > > and then try to unlock again.
> > 
> > 
> > Thanks my friend , can you teach me how to make this apk? can make it by myself?
> 
> It's very complex and will use many tools, the main steps are:
> 
> * get the `payload.bin` file from the earliest HyperOS ROM.
> * get the `system_ext.img` from `payload.bin`
> * get the apk file from `system_ext.img`

Thank you for the detailed instructions. However, how can I determine which version is the correct one and the one I need?

blackmatch, 2024-09-30:
[image]

> how can I determine which version is the correct one and the one I need?

check this code on your mobile.

Smartslabs, 2024-09-30:
> ![image](https://private-user-images.githubusercontent.com/12443954/371926269-df892714-71ea-426d-aafc-383056e8e166.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc2Njk1NjcsIm5iZiI6MTcyNzY2OTI2NywicGF0aCI6Ii8xMjQ0Mzk1NC8zNzE5MjYyNjktZGY4OTI3MTQtNzFlYS00MjZkLWFhZmMtMzgzMDU2ZThlMTY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTMwVDA0MDc0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUwOTQ4ZDQyYTkyYTlkNzllMDA3Yjc0Y2Q3YWFiMDBiNmEwMzJmNGE4Nzg3ODc1OGIzNTc2NzQ2YjdjNGM1YWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mCQdSZmlRAQ3efDuFQX4uf7QOkO_n3El-xICQrkMyLE)
> 
> > how can I determine which version is the correct one and the one I need?
> 
> check this code on your mobile. the ROM file name includes this code, for example: <img alt="image" width="1227" src="https://private-user-images.githubusercontent.com/12443954/371927567-01a1ddae-26ae-4b05-9f52-c32f280f64ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc2Njk1NjcsIm5iZiI6MTcyNzY2OTI2NywicGF0aCI6Ii8xMjQ0Mzk1NC8zNzE5Mjc1NjctMDFhMWRkYWUtMjZhZS00YjA1LTlmNTItYzMyZjI4MGY2NGVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTMwVDA0MDc0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxYjA0ZjY4MWM5MzJhNzE1NWJmMDMxNWIwOGY5Y2NmYWNhN2M3NGNjYmU2NjVhMjJlM2U2ZGRmYmM3MGE4ZDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FM1UGfA9NTMOu4AiT3ZbWXPdkjWkFG4W3YzqLxFUP7M">

Same as my existing OS? So why should install it ?

keenan316, 2024-09-30:
Mine is HyperOS 1.0.16.0 UNREUXM and non of the settings.apk can be installed to downgrade.

BlazingRiser, 2024-09-30:
You need to install the oldest version of settings apk from hyperos that is 1.0.1.0 followed by your region code matching your about phone page. Also check if bypass is supported or not in check.csv, under main files on github.(X6 5g is redmi note 13 pro, codename garnet)

code4fun1024, 2024-09-30:
Thanks to @blackmatch , I finally extract the correct apk file. The apk file is from HyperOS 1.0.8.0 `UNREUXM` : https://drive.google.com/file/d/18VGV58Ci9eCs4TgR2ZCNdtAhE0fS5tIG/view?usp=sharing

@keenan316 you can try this apk file.

keenan316, 2024-09-30:
> Thanks to @blackmatch , I finally extract the correct apk file. The apk file is from HyperOS 1.0.8.0 `UNREUXM` : https://drive.google.com/file/d/18VGV58Ci9eCs4TgR2ZCNdtAhE0fS5tIG/view?usp=sharing
> 
> @keenan316 you can try this apk file.

Hi @code4fun1024

I just tried it, but unfortunately it's still the same "App not installed as package appears to be invalid"

Thanks for your effort though..

vuccahh, 2024-09-30:
> > App not installed as package appears to be invalid
> 
> I guess I used the wrong ROM to extract the apk file, can anyone tell me what the ROM are you use ? so that I can extract the correct apk file. The Global one ?

sorry for bothering you , i have xiaomi poco pad (redmi pad) using 1.0.7.0 UNSMIXM
getting error 10000 , tried everything out there but no luck, just asking if its really rootable?

AlnorFromYT, 2024-09-30:
@blackmatch please help, I need the correct settings app for mine🙁, I'm using XIAOMI 11T  using HyperOS at 1.0.6.0.UKWMXM please🥹

Smartslabs, 2024-10-04:
No anyone can provide  Note 13 Pro / Poco x6 5G update setting.apk?

tbprothai, 2024-10-08:
錯誤30001，我應該點做？

code4fun1024, 2024-10-09:
https://github.com/code4fun1024/setting-app-for-hyperos

@vuccahh I've extract the apk file from OS1.0.1.0.UNSMIXM.
@AlnorFromYT your "ROM code" should be `UKWMIXM` ? I extract the apk file from OS1.0.2.0.UKWMIXM.

anyone who want the old version Setting app apk, please comment your "ROM code" [here](https://github.com/code4fun1024/setting-app-for-hyperos/issues/1)

moffa13, 2024-11-05:
For those who need the specific Settings.apk for their ROM code
Download the oldest hyperos recovery rom (zip file, not fastboot)
extract the zip and get payload.bin
use a tool like extract_android_payload to extract system_ext.img (python extract_android_payload.py extract -p system_ext payload.bin /tmp/)

extract the contents using a tool like erofsUnpack
Search for Settings.apk

Tigercat000, 2024-11-05:
> For those who need the specific Settings.apk for their ROM code Download the oldest hyperos recovery rom (zip file, not fastboot) extract the zip and get payload.bin use a tool like [...]

oldest hyperos recovery rom?that's mean is the first out hyper version ? Thanks

moffa13, 2024-11-05:
> oldest hyperos recovery rom?that's mean is the first out hyper version ? Thanks

I'm not sure but I guess yes, I was not able to test as my hyperos version (1.0.8.0) does not allow me to update settings app

BlazingRiser, 2024-11-15:
> I'm not sure but I guess yes, I was not able to test as my hyperos version (1.0.8.0) does not allow me to update settings app

Check if bypass is supported or not for your device first if settings apk is not installing [here](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/blob/master/status.md)

bdd786, 2025-01-29:
can someone send the earlier version of settings.apk for UNRINXM?

Yunxmi, 2025-02-01:
Settings apk for garnet, hyperos 2.0.3.0 VNREUXM

soul686, 2025-05-04:
> 由于[@blackmatch](https://github.com/blackmatch)终于，我提取到了正确的 apk 文件。该 apk 文件来自[HyperOS](https://drive.google.com/file/d/18VGV58Ci9eCs4TgR2ZCNdtAhE0fS5tIG/view?usp=sharing)`UNREUXM` 1.0.8.0：https://drive.google.com/file/d/18VGV58Ci9eCs4TgR2ZCNdtAhE0fS5tIG/view?usp=sharing[](https://drive.google.com/file/d/18VGV58Ci9eCs4TgR2ZCNdtAhE0fS5tIG/view?usp=sharing)
> 
> [@keenan316](https://github.com/keenan316)你可以尝试这个apk文件。

能讲述一下怎么提取的旧版设置吗