MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #184: Can't run file CMD to bypass

ChitatoGM, 2024-09-29:
The problem is i can't even run the Bypass.cmd file. Every time i run with adminstrator, the window of cmd just close immediately. I'm tried to fix it but nothing is worked. Is there a problem with my computer?

BlazingRiser, 2024-09-30:
Followed instructions given on main page?, if so try running bypass.php instead of bypass.cmd.

ChitatoGM, 2024-10-01:
> Followed instructions given on main page?, if so try running bypass.php instead of bypass.cmd.
>I thought it was just download the PHP file and run it? What else do I need to do? I watched some videos on Youtube and they just said to download the PHP file and run it.

ChitatoGM, 2024-10-01:
> Followed instructions given on main page?, if so try running bypass.php instead of bypass.cmd.
> 

I thought I just needed to download the PHP file and run the cmd file? I saw some videos on Youtube saying I just needed to download the php file and run it.

stevleibelt, 2024-10-01:
@ChitatoGM Just follow the official [instruction](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass?tab=readme-ov-file#%EF%B8%8F-how-to-use) and not a youtube video :-).

Fingers crossed this helps. If not, add your command line output in here.

ChitatoGM, 2024-10-01:
> @ChitatoGM Just follow the official [instruction](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass?tab=readme-ov-file#%EF%B8%8F-how-to-use) and not a youtube video :-).
> 
> Fingers crossed this helps. If not, add your command line output in here.

I'm not understanding the tutorial part of it because everything is a bit abstract and there are no specific pictures or videos to work with:)))

Bemzter12, 2024-11-24:
Update your microsoft visual c++