MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #185: Old version of Settings app for Redmi 12C

su1cidalma9iac, 2024-09-29:
hi, I'm getting the "Request Parameter Error (10000)" error while trying to unlock my Xiaomi Redmi 12C. I need the older version of the settings app if anyone can provide me. Or maybe tell me a way to get it from somewhere. I can not find it on Google or any forums.

Marvin732, 2024-10-17:
Did you try the Settings app that is in the zip when you go to Releases and download the zip?

su1cidalma9iac, 2024-10-22:
yes. it didn't install. and anyways, i had to take the hard bricked phone to the service center in another city and they flashed miui 13 (Android 12) for me and it binded account with the device without any problems this time and i have applied for unlocking. they gave me 7 days wait time.

MobySensei, 2024-10-22:
> yes. it didn't install. and anyways, i had to take the hard bricked phone to the service center in another city and they flashed miui 13 (Android 12) for me and it binded account with the device without any problems this time and i have applied for unlocking. they gave me 7 days wait time.

Bro...can u plz..help me .....i have the same device "Redmi 12C" ..and it giving me the same "invalid package" error whenever i tried to to update the "setting" app ....can plz u guide me?...about how to bypass bootloader?.....( I would be very grateful to you....)....

su1cidalma9iac, 2024-10-22:
i'd suggest the official method if you can do it. apply for permissions through the xiaomi community app. if your account is not 30 days old then wait for it to be. i'm saying this because i don't want you to end up with a bricked phone with locked bootloader that can't be flashed without some premium tool. save yourself some trouble and money.

MobySensei, 2024-10-22:
> i'd suggest the official method if you can do it. apply for permissions through the xiaomi community app. if your account is not 30 days old then wait for it to be. i'm saying this because i don't [...]

Hmm.....bro i've tried their community app .. and whenever i apply for bootloader they say "qouta limit reached" and gave some time to try again later ....i also created another fake mi account with US religion and have waited 30 days ...( I thought ...in this way .. it'll probably gonna work ...)....but ..all in vain ......

su1cidalma9iac, 2024-10-22:
hmm i see! i don't have any way to flash previous versions of miui. i got it done from a service center because they have all the premium tools which get it done. have you tried downgrading settings app with adb?

iamtheabdi, 2024-10-24:
try

```adb install -r -d App.apk```

but I think this might fail since it's a system app. it can't be downgraded without root which again requires you to unlock the bootloader and you're doing all this to unlock the bootloader already.

closed 2024-10-24 by iamtheabdi