MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #188: Fail to send request, check your internet connection

Baglanoc, 2024-10-07:
[image]
Please, help me! 
How to solve this problem?

Baglanoc, 2024-10-07:
Redmi note 13 pro 5g

Baglanoc, 2024-10-07:
Chinese version

atingee, 2024-10-09:
Turn on USB tethering, and you can see the Android phone in the device manager（adb）. Do not connect to the network through the Xiaomi phone, but use another network (another mobile phone or a direct connection via a network cable). Then, in Network adapters, disable the network card of this phone. Run the unlock command again.

Xssgit47, 2024-10-11:
> Turn on USB tethering, and you can see the Android phone in the device manager（adb）. Do not connect to the network through the Xiaomi phone, but use another network (another mobile phone or a direct [...]

But it's Require sim card internet, and when i click on "add account and device", internet connection goes off automatically, that's why we got the post request error. can you solve the internet issue?

cced1900, 2024-10-12:
When I was just using the Xiaomi and accessing the Internet via USB sharing, I had the same situation you have now. It shows network problems. Because the Xiaomi is in flashboot state, it is impossible to network. So you need another network (wired, wireless or another hotspot).

> But it's Require sim card internet, and when i click on "add account and device", internet connection goes off automatically, that's why we got the post request error. can you solve the internet [...]

atingee, 2024-10-15:
> But it's Require sim card internet, and when i click on "add account and device", internet connection goes off automatically, that's why we got the post request error. can you solve the internet [...]

What is your network environment like?
My computer becomes inaccessible after running the Bypass batch command with adb when only my phone is connected to the Internet.
If I use another phone as a hotspot to access the Internet, and inside the network adapter, I disable the USB shared network card of my Xiaomi phone, and the phone's network just remains in normal use. The Xiaomi phone is successfully connected after running the Bypass batch command with adb in ADB state.
BTW, what model is your phone? It seems that Redmi Note 13 Pro is not supported.
Xiaomi-HyperOS-BootLoader-Bypass/status.md on master - MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass (github.com)

Xertien, 2024-11-13:
I finally found the problem. You have to be using a USB 2.0 port. Disable and re-enable your mobile data. It worked fine for me and resolved my problem

Xssgit47, 2024-11-13:
Thanks for helping me,

      In my case, i turned the OEM off and than turned on it quickly. This
process solved my problem.