MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #192: Help needed !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

hungrymachine1, 2024-10-08:
[image]
Ihave downloaded the php file but i dont know how to follow the instructions or how to do these. Can you provide an brief tutorial about it

Sudharsshan, 2024-10-19:
Well, it's relatively simple to follow:

1. Extract the downloaded .zip file
2. you'll find the **php.ini-development** file
3. Rename that file to **php.ini**
4. Open it in any text editor like Notepad and find 'extension' using ctrl+f
5. next remove the semi-colon at the starting of the lines where it says **extension=curl** and **extension=openssl**
6. Save it.
7. Next hit ctrl+f and find **extension_dir** (there are two of them, uncomment the lower one by removing the starting semi-colon)
8. Save it again.
9. Add this folder into system variables under the 'PATH' by following any simple online guide
10. then open your command prompt and type:  `php <path_to_bypass.php_file>`
11. next hit enter and connect your device.

It should work, if not pls let me know

shenkunlovecoding, 2025-01-01:
If you still haven't configured it properly
Try using the one click Python version https://pan.baidu.com/s/1h-ok3CH3mFyMg4NxcWm3og?pwd=arap