MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #194: bypass worked with downgraded settings, unlock tool still fails

murdegern, 2024-10-11:
Hi all and let me thank you first for your amazing work which may enable me to still liberate my Xiaomi 11 Lite 5G NE (lisa).

After following the existing issues, I downgraded my Settings app to Settings_AGATEGlobal_from_OS1.0.2.0.UKWMIXM and hooray - the bypass worked:

[2024-10-11] [11:33:23] [INFO] - Account bound successfully, wait time can be viewed in the unlock tool.

... but this does not work. The unlock tool goes to 99%, then shows "This device is not activated, please activate it and try to unlock it again." :-(

Can anyone help me with this?

Thanks a lot in advance!

jwtiyar, 2024-12-19:
How you did downgrade the settings app?

murdegern, 2024-12-19:
with adb - very quick overview:
you need to install adb (android debug bridge) on your PC, activate developer mode on your phone (Settings -> About phone -> tap "build number" 7 times), activate usb debugging, connect phone and PC and accept the debugging.
Then you can "adb install -r -d your-new-settings.apk"

jwtiyar, 2024-12-19:
> with adb - very quick overview: you need to install adb (android debug bridge) on your PC, activate developer mode on your phone (Settings -> About phone -> tap "build number" 7 times), activate usb [...]

I know bro, but it doesnt't work this way i thought you tried something new to these their ROM version is newer.