MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #200: The place where the account is registered does not match the place where the phone is sold. (20045)

visionblurry, 2024-10-17:
that's issue was at xiaomi pad 6 from china.
i can use bypass and mi unlock take me 168 hours, but from 7 days i take this error in mi unlock tool.

M2Play, 2024-10-18:
It seems that the new code 20045 replaced the old 30001 for Chinese devices.. Xiaomi has again changed the BL unlocking conditions and the bypass is currently not working.

Razzyismness, 2024-10-21:
I've been trying to unlock the bootlocker of my Xiaomi Redmi Turbo 3 for hours. And I can't seem to do it no matter how many videos i watch or instructions i follow. Someone help me please with a viable solution to this damn problem.

xiaohuzz, 2024-10-22:
For now, only Chinese accounts can solve this problem

cute-omega, 2024-12-29:
#236

Ghostly938, 2025-04-18:
Make a chinese account. You don't need a chinese number to do it. Create a new account, set the region to china then put it your unused sim card number

hubertwalter05-eng, 2026-10-21:
> 6\. Créez un compte chinois. Vous n'avez pas besoin d'un numéro chinois. Créez un nouveau compte, sélectionnez la Chine comme région, puis indiquez le numéro de votre carte SIM inutilisée.

C'est pas possible maintenant !