MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #201: Can anyone help - ROM - Poco x6 5g - 1.0.8.0.UNPMIXM

betamaster8, 2024-10-19:
Hi guys,
a ton of thank for putting up this fantastic project - kudos kudos kudos

[yyyy-mm-dd] [hh:mm:ss] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in {Drive}\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in {Drive}\php\bypass.php on line 333
[yyyy-mm-dd] [hh:mm:ss] [INFO] - Sending POST request...
[yyyy-mm-dd] [hh:mm:ss] [INFO] - Request parameter error (10000)

ROM - Poco x6 5g - 1.0.8.0.UNPMIXM
ROM Link
https://xmfirmwareupdater.com/hyperos/peridot/stable/OS1.0.8.0.UNPMIXM/

Thanks guys
@MlgmXyysd @s1204IT @stevleibelt

stevleibelt, 2024-10-19:
@betamaster8 How do you use this tool?

With [my pull request](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/203), I've added a check that stops executing the script if an expected key does not exist in the `$data`.
But this fix does not solve your problem right now. Maybe you can comment the error message that is printed if you try it with my pull request.

Kind regards,
stev

betamaster8, 2024-10-19:
Thanks @stevleibelt 

Getting following Error

PHP Parse error:  syntax error, unexpected token "}" in {Drive}\bypass.php on line 237
Parse error: syntax error, unexpected token "}" in {Drive}\bypass.php on line 237

Using PHP and Adb on terminal
Cheers
Bm

stevleibelt, 2024-10-21:
As @betamaster8 has written down [here](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/203#issuecomment-2425280749), he now has an issue I hope @MlgmXyysd or @s1204IT can fix.

> Thanks @stevleibelt
> 
> 
> PHP Fatal error: Uncaught TypeError: array_key_exists(): Argument https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/2 ($array) must be of type array, null given in {Drive}bypass.php:338
> Stack trace:
> #0 {main}
> thrown in {Drive}bypass.php on line 338
> 
> Fatal error: Uncaught TypeError: array_key_exists(): Argument https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/2 ($array) must be of type array, null given in {Drive}bypass.php:338
> Stack trace:
> #0 {main}
> thrown in {Drive}bypass.php on line 338
> 
> and it turns off data on the SIM, kind of weird behavior by HyperOS/Xiaomi ???
> Cheers

This is now an error kind of expected. `$data` is not an array so there must be an issue in the `$args`. But that is the part I totally have no clue about. What I did so far was to add checks if `$data` is looking as expected.

betamaster8, 2024-10-23:
Awesome mates @stevleibelt 

Looking forward to solve the issue of array data with @MlgmXyysd and @s1204IT 

Cheers
Bm

vpolinov, 2024-12-01:
I have the same issue - "Request parameter error (10000)" with my poco x6 5g
Unfortunately, reading all the posts here I cannot just understand if there is any possible way out? No updates since Oct 23...
Can someone comment on this, please?

cute-omega, 2024-12-29:
#236