MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #202: Code is not detecting my phone

Sudharsshan, 2024-10-19:
I carried out the operations as exactly given in the instructions, but in the command prompt it just doesn't detect my phone or whatever?

I have **Redmi 13C 5G**, which i recently updated into HyperOS (seriously why did I even do that..)

This is what it shows in the command prompt when I execute the script:
<br>
[image]
<br><br>
Even after unplugging out the phone and plugging it back in, it just doesn't get detected by the script. 
<br><br>
Any help is appreciated, Thanks!

BlueNoob, 2024-10-20:
phone should be in FAST BOOT mode.

Sudharsshan, 2024-10-23:
Thank you though I tried, it did not detect as the code requires you to unlock and connect the phone so that it can take you to the developer options menu.

MihailoDevelops, 2024-10-23:
Have you turned Windows real-time protection off? I had the same problem until I turned it off and then it started detecting the phone.

noicl, 2024-11-29:
You need the adb.exe in your libraries folder. The libraries folder needs to be in the same folder as the bypass.php script

Sudharsshan, 2024-11-30:
Hey thanks for your help, I tried it out yet it just won't detect my phone. I have just dropped it at this point and moved on to Samsung. 
Appreciate it guys thanks!

closed 2024-11-30 by Sudharsshan