MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #204: settings.apk doesn't work

BlueNoob, 2024-10-20:
App not installed as package appears to be invalid.
[image]

BlueNoob, 2024-10-20:
[image]

betamaster8, 2024-10-23:
@BlueNoob - The settings app is old version
While the updated HyperOS cannot install older settings app on to it

You need to downgrade the settings app
That is only possible after root

But root is only possible after unlocking bootloader
Cheers
Bm

marufmoinuddin, 2024-10-25:
> @BlueNoob - The settings app is old version While the updated HyperOS cannot install older settings app on to it
> 
> You need to downgrade the settings app That is only possible after root
> 
> But root is only possible after unlocking bootloader Cheers Bm

so there is no way to unlock bootloader? will xiaomi community bootloader unlock method still work?

betamaster8, 2024-10-30:
Seems the OP of this project @MlgmXyysd is not interested to maintain this project anymore :(

cute-omega, 2024-12-29:
maybe you can find something useful here #236