MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #209: XIOMI Pad 6 || Error - Request parameter error (10000)

nalindev, 2024-10-30:
why i am getting this error. someone please help me -
```
Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333
[2024-10-30] [14:12:49] [INFO] - Sending POST request...
[2024-10-30] [14:12:50] [INFO] - Request parameter error (10000)
```

[image]

nalindev, 2024-10-30:
Can someone fix this issue? As @stevleibelt mentioned in the merge req, the returned data is not an array.

his merge request is also not working for me; it gets stuck after reaching the point shown in the screenshot and doesn’t progress further.

I believe the data might be in encrypted form. If so, is there a way to decrypt it?

Please let us know if there are any possible solutions to patch this issue. Thanks in advance!

[image]

nalindev, 2024-10-30:
### **ISSUE FIXED**

The issue is finally fixed for me, and I see that many of you are experiencing the same problem. I wanted to let you know that you also need to install the settings APK from this repo before running the bypass cmd file.

After doing this, it’s working for me.

[image]

biggestlong, 2024-11-01:
我的手机安装不了旧版的设置怎么解决这个问题0？

YSDWR, 2024-11-08:
> 我的手机安装不了旧版的设置怎么解决这个问题0？

具体怎么安装不了？如果是提示无法从非官方渠道安装系统应用的话，去开发者选项最下面 启用系统优化 关掉就可以了

Rizado, 2024-11-24:
@nalindev , please share your solution of this problem.

ricefirst, 2024-12-04:
> > 我的手机安装不了旧版的设置怎么解决这个问题0？
> 
> 具体怎么安装不了？如果是提示无法从非官方渠道安装系统应用的话，去开发者选项最下面 启用系统优化 关掉就可以了

显示「应用未安装：软件包似乎无效」

martinezix00, 2025-01-27:
> ### **ISSUE FIXED**
> The issue is finally fixed for me, and I see that many of you are experiencing the same problem. I wanted to let you know that you also need to install the settings APK from this repo before running the bypass cmd file.
> 
> After doing this, it’s working for me.
> 
> [image]

Share the solution with others please.

nalindev, 2025-01-27:
> Share the solution with others please.

Martin you need to install the "settings apk" from this repo before running the bypass cmd file.

stevleibelt, 2025-01-28:
@nalindev what do you mean with "settings apk"? Maybe we can improve the [README](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass#) if something is missing there.
Do you refer to the [official unlock tool](https://en.miui.com/unlock/index.html) maybe?

nalindev, 2025-01-28:
@stevleibelt go to releases and check the file. There is a apk attached with it.

martinezix00, 2025-01-28:
> > > ### **ISSUE FIXED**
> > > The issue is finally fixed for me, and I see that many of you are experiencing the same problem. I wanted to let you know that you also need to install the settings APK from this repo before running the bypass cmd file.
> > > After doing this, it’s working for me.
> > > [image]
> > 
> > 
> > Share the solution with others please.
> 
> Martin you need to install the "settings apk" from this repo before running the bypass cmd file. Otherwise it will not work.

I've tried different Settings.apk files but there's no option to install Settings app dedicated for Android 14 on Android 15 device. My device is Pad 6S CN version with HyperOS 2.0.3.0 and I'm afraid that currently there's no explotable version to use the patch. By the way - I cannot find neither apk nor releases section

cow-your-sister, 2025-02-08:
> I've tried different Settings.apk files but there's no option to install Settings app dedicated for Android 14 on Android 15 device. My device is Pad 6S CN version with HyperOS 2.0.3.0 and I'm [...]

Same. I'm using Pad 6 Pro CN on HyperOS2.0.4.0, it's based on Android 15, so the previous settings application might not be installed successfully. How I wish if I had not upgraded to OS2.0! If there're any ways to solve this confliction? Although I had reached level 5 in Mi Community, the nearest bootloader-unlock-exam is going to be hold on Feb. 18th the Twesday, right 2 days after our now school term started😭. That's too bad.

blackshark555666, 2025-02-10:
> I've tried different Settings.apk files but there's no option to install Settings app dedicated for Android 14 on Android 15 device. My device is Pad 6S CN version with HyperOS 2.0.3.0 and I'm [...]

same here 😭😭
xiaomi more worst

kaif-00z, 2025-03-21:
> ### **ISSUE FIXED**
> The issue is finally fixed for me, and I see that many of you are experiencing the same problem.
> 
> Solution 👉 you also need to install the `settings APK` from this repo before running the bypass cmd file.
> 
> After doing this, it’s working for me.
> 
> [image]

can you give the link of apk?

MeaningOrca, 2025-06-26:
So does anybody find the solution for it? My Xiaomi pad 6 with HyperOS 2.0.8.0, how do I install the settings? When I'm trying through adb I've got:
```
C:\platform-tools>adb install Settings.apk
Performing Streamed Install
adb: failed to install Settings.apk: Failure [INSTALL_FAILED_INVALID_APK: Package com.android.settings are not updateable.]
```