MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #215: Resolving timed out

callmezeemka, 2024-11-05:
When i try to run it for some reason it always gave me "Fail to send request, check your internet connection" so i changed the code slightly to see where problem is and it looks like when it does requests it gives me error "Resolving timed out after 6 seconds" which is strange because i can access the domain in my website

Phone redmi 12 
[image]

callmezeemka, 2024-11-05:
so i changed the original url that was in code to extranet-ipv4-pub-alisgp.alb.xiaomi.com (this one showed up when i did dns lookup) and now it shows OpenSSL/3.0.15: error:0A000410:SSL routines::sslv3 alert handshake failure error

callmezeemka, 2024-11-05:
okay i got it it was my dns problem ill check if it works through fastboot and close if it worked

callmezeemka, 2024-11-05:
[image]
yep looks like it works

closed 2024-11-05 by callmezeemka

stevleibelt, 2024-11-06:
@callmezeemka If you want to, you can create a pull request.

Either by replacing the url as you did or by implementing a logic that iterates over an array of url's and tries the next one if the first one is not working.

Thanks for letting us know how you've fixed the issue.