MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #217: ADB已经连接了，但是脚本Waiting for device connection...

Canmi21, 2024-11-09:
[image]
[image]
[image]
[image]
症状是，手机小米12S，开了ADB调试 OEM解锁，ADB安全调试，插上电脑资源管理识别了
[image]
但是似乎一直卡在等待ADB连接这个界面，开始怀疑是我的问题，但是使用了秋之盒查看
[image]
结果显示手机已连接，然后使用ADB查看
[image]

closed 2024-11-12 by Canmi21

kiwirafe, 2024-12-20:
您好 请问你是怎么解决的啊？

MeaningOrca, 2025-06-26:
Same problem. Does somebody find the solution?