MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #219: CMD窗口报错，麻烦看看问题

951240778, 2024-11-12:
请问一下这个报错什么原因，想绕过设备解锁，谢谢答复
[image]

DarkDuck007, 2024-11-15:
Yes, I am getting the same error.
Apparantly decryptData returns and empty value after trying to decode $args even though $args itself isn't empty.

Mostwest, 2024-12-07:
> Yes, I am getting the same error. Apparantly decryptData returns and empty value after trying to decode $args even though $args itself isn't empty. I also can't seem to get the overridden settings.apk to work. EDIT: I got it to work with a simple adb install settings.apk then do everything as told, no error 1000 anymore.

Which device and which update you are referring to?

i've read people that said it wasn't working for them, I'm pretty sure they where referring to HyperOs devices with recent updates