MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #221: fatal error?

Radiantguled, 2024-11-17:
[image]
i keep getting this error. i put the file in the c drive and ran it as administrator. Was their a new update to the hyper os?

Madin-David, 2024-11-24:
This is caused by curl and openssl not being loaded
As you can see from the first few wanrning
The solution is to remove the ; before ;extension_dir=“ext” (line 768) in php.ini

ginolaginola, 2024-12-27:
> This is caused by curl and openssl not being loaded As you can see from the first few wanrning The solution is to remove the ; before ;extension_dir=“ext” (line 768) in php.ini

I delete the ;extension_dir=“ext” , but the same problem.
When I want to add Mi accont in the developer option the mobile interenet closed, why? 

PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\Miklos Tibor\Downloads\27\bypass.php:230
Stack trace:
#0 C:\Users\Miklos Tibor\Downloads\27\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\Users\Miklos Tibor\Downloads\27\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\Miklos Tibor\Downloads\27\bypass.php:230
Stack trace:
#0 C:\Users\Miklos Tibor\Downloads\27\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\Users\Miklos Tibor\Downloads\27\bypass.php on line 230

ginolaginola, 2025-01-01:
> Try using the Python version https://pan.baidu.com/s/1h-ok3CH3mFyMg4NxcWm3og?pwd=arap

Hi. Thanks.
What is this exe file?

shenkunlovecoding, 2025-01-04:
The Python rewrite version of this repository, with the source code stored in my repository

Zxingtsai, 2025-01-25:
Similar on MacOS
[2025-01-25] [15:31:55] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php on line 330 and defined in /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php:228
Stack trace:
#0 /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php(330): decryptData(NULL)
#1 {main}
  thrown in /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php on line 228

Fatal error: Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php on line 330 and defined in /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php:228
Stack trace:
#0 /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php(330): decryptData(NULL)
#1 {main}
  thrown in /Users/jiangyunqin/Downloads/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php on line 228

Madin-David, 2025-01-26:
> I delete the ;extension_dir=“ext” , but the same problem. When I want to add Mi accont in the developer option the mobile interenet closed, why?
>
> PHP Fatal error: Uncaught Error: Call to undefined [...]

Try not delete  ;extension_dir=“ext”
delete the ; before  ;extension_dir=“ext”
that is to say to change  ;extension_dir=“ext” into  extension_dir=“ext”