MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #222: bypass fails

MasterDickLOL, 2024-11-26:
C:\php>bypass.cmd
[2024-11-26] [06:42:32] [INFO] - ************************************
[2024-11-26] [06:42:32] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-11-26] [06:42:32] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-11-26] [06:42:32] [INFO] - ************************************
[2024-11-26] [06:42:32] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-11-26] [06:42:32] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-11-26] [06:42:32] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-11-26] [06:42:32] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-11-26] [06:42:32] [INFO] - My Blog: https://www.neko.ink/
[2024-11-26] [06:42:32] [INFO] - ************************************
[2024-11-26] [06:42:32] [INFO] - Starting ADB server...
[2024-11-26] [06:42:33] [INFO] - Processing device c1821a36(6)...
[2024-11-26] [06:42:33] [INFO] - Finding BootLoader unlock bind request...
[2024-11-26] [06:42:33] [INFO] * Now you can bind account in the developer options.
[2024-11-26] [06:43:38] [INFO] - Account bind request found! Let's block it.
[2024-11-26] [06:43:38] [INFO] - Refactoring parameters...
[2024-11-26] [06:43:38] [INFO] - Sending POST request...
[2024-11-26] [06:43:42] [INFO] - The place where the account is registered does not match the place where the phone is sold. (20045)

model: redmi k70 pro ALSC
HyperOS version: 1.0.22.0.UNMCNXM
Account registered in Iraq
phone bought online from china

mducng697, 2024-11-26:
I think your device was activated online by the store in China

MasterDickLOL, 2024-11-26:
> I think your device was activated by the store in China

So any way to unlock bootloader without having a Chinese mi account?

MasterDickLOL, 2024-11-26:
> Currently, there's no way.

And is there a way to get a Chinese Xiaomi account?