MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #223: error php bypass

Renger1951, 2024-11-28:
[2024-11-28] [22:12:53] [INFO] - Account bind request found! Let's block it.
[2024-11-28] [22:12:53] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in C:\php\php\bypass.php:230
Stack trace:
#0 C:\php\php\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\php\php\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\php\php\bypass.php:230
Stack trace:
#0 C:\php\php\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\php\php\bypass.php on line 230

Dephrilibrium, 2024-12-15:
The message tells you that there is no openssl. Note that when you running on windows, you need to uncomment the setting `extension_dir = "ext"` (line 768) setting in `php.ini`.

ginolaginola, 2024-12-27:
> Az üzenet azt mondja, hogy nincs openssl. Ne feledje, hogy amikor Windowson futtatja, törölnie kell a beállítás `extension_dir = "ext"`(768. sor) beállításának megjegyzését a -ban `php.ini`.

It is not work, the sáme mistake, than above.

ginolaginola, 2024-12-27:
> [2024-11-28] [22:12:53] [INFO] - Számlakötési kérelem található! Letiltjuk. [2024-11-28] [22:12:53] [INFO] - Paraméterek újrafaktorálása... PHP Végzetes hiba: Uncaught Error: Undefined függvény hívása openssl_decrypt() a C:\php\php\bypass.php:230-ban Veremkövetés: #0 C:\php\php\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...') #1 {main} a C:\php\php\bypass.php fájlban a 230-as sorban
> 
> Végzetes hiba: Nem elkapott hiba: Az openssl_decrypt() nem definiált függvény hívása a C:\php\php\bypass.php:230 Veremkövetés: #0 C:\php\php\bypass.php(330): decryptData('8/) 8Mnp7WcRph8sb...') #1 {main} bedobva C:\php\php\bypass.php a 230-as sorban

Similar...Can you solve the problem?

Starting ADB server...
[2024-12-27] [13:03:39] [INFO] - Processing device 9D4XAYIVUKIBW86T(12)...
[2024-12-27] [13:03:40] [INFO] - Finding BootLoader unlock bind request...
[2024-12-27] [13:03:40] [INFO] * Now you can bind account in the developer options.
[2024-12-27] [13:04:03] [INFO] - Account bind request found! Let's block it.
[2024-12-27] [13:04:03] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\Miklos Tibor\Downloads\27\bypass.php:230
Stack trace:
#0 C:\Users\Miklos Tibor\Downloads\27\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\Users\Miklos Tibor\Downloads\27\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\Miklos Tibor\Downloads\27\bypass.php:230
Stack trace:
#0 C:\Users\Miklos Tibor\Downloads\27\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\Users\Miklos Tibor\Downloads\27\bypass.php on line 230

stevleibelt, 2024-12-27:
@ginolaginola and @Renger1951,

as found in the [first two search hits](https://duckduckgo.com/?t=ffab&q=php+openssl+windows&ia=web).

> 1. Copy php.ini-development to php.ini
> 2. Uncomment the extension=php_openssl.dll line as described
> 3. Uncomment this line: extension_dir = "ext"
> 4. Make sure you find php_openssl.dll within that extension directory of course

[ source](https://superuser.com/a/1051443)

As an alternative, [this](https://www.herongyang.com/PKI/HTTPS-PHP-Configure-PHP-OpenSSL-on-Windows.html) is looking promising ... written from a Linux/FreeBSD user who has almost no clue about Windows.

ginolaginola, 2024-12-27:
> @ginolaginola and @Renger1951,
>
> as found in the [first two search hits](https://duckduckgo.com/?t=ffab&q=php+openssl+windows&ia=web).
>
>
> [ source](https://superuser.com/a/1051443)
>
> As an alternative, [...]

Thanks, the same message.
When I add in the developer option my Mi account, the mobile net is off....

Dephrilibrium, 2024-12-28:
> > Az üzenet azt mondja, hogy nincs openssl. Ne feledje, hogy amikor Windowson futtatja, törölnie kell a beállítás `extension_dir = "ext"`(768. sor) beállításának megjegyzését a -ban `php.ini`.
> 
> It is not work, the sáme mistake, than above.
> I delete the `extension_dir = "ext", but not solved...

Not deleting. Just uncommenting, so that the line is not ignored.

stevleibelt, 2024-12-28:
@ginolaginola where you able to verify that the openssl library is loaded? A common way is [phpinfo](https://www.php.net/manual/en/function.phpinfo.php). You could call `php -i`. Under Unix, you could use `php -i | grep -i ssl` but I have no idea how to do this under Windows.
Before you could verify that OpenSSL is loaded, you should not start any new unlocking round.

ginolaginola, 2024-12-28:
> @ginolaginolahol tudod ellenőrizni, hogy az openssl könyvtár be van-e töltve? Egy általános módszer a [phpinfo](https://www.php.net/manual/en/function.phpinfo.php) . Felhívhatnád `php -i`. Unix alatt használhatod, `php -i | grep -i ssl`de fogalmam sincs, hogyan kell ezt megtenni Windows alatt. Mielőtt ellenőrizné, hogy az OpenSSL be van-e töltve, ne kezdjen új feloldási kört.

Now:
Binding failed with error code `30001

shenkunlovecoding, 2025-01-01:
Try using the Python version