MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #224: line 230 Fatal error

Game-ger, 2024-11-29:
是我新版php问题吗
[image]

Game-ger, 2024-11-29:
等待解锁的是redmi turbo3

vpolinov, 2024-12-01:
Same story here.
It's poco x5 5G global version running 1.0.12.0.UNRIDXM
[image]

Game-ger, 2024-12-04:
不过你的机器应该可以解锁，看报错应该是依赖的包没有启用

Hhehepei, 2024-12-04:
你好，请检查一下php.ini中的extension_dir这一配置项，确保它指向了php拓展文件的所在路径，在windows上这应该是php安装路径中的ext文件夹

bingdianfeishui, 2024-12-06:
> 你好，请检查一下php.ini中的extension_dir这一配置项，确保它指向了php拓展文件的所在路径，在windows上这应该是php安装路径中的ext文件夹

感谢提醒，Redmi K40已成功解锁。报错时前面有提示找不到"C:\php\ssl\xxxx"动态库。需要将php.ini中以下内容解除注释：
```
extension_dir = "ext"
```