MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #226: Did Xiaomi fixed this vulnerability?

yipeyayeah, 2024-12-09:
Did Xiaomi fixed this vulnerability?
After successfully sending the POST request, the request returned Error Code of 20045, which is the same error for every generic HyperOS device trying to unlock.
[image]

mducng697, 2024-12-13:
Just use a Xiaomi account with the China region.

cute-omega, 2024-12-29:
#200

Ghostly938, 2025-04-18:
Create a new xiaomi account with the region set to china, no need to vpn or a chinese number. You can use your own number

ahzay, 2025-09-26:
> no need to vpn or a chinese number

When I select the "China" region, it only allows +86 phone number signups, no email. It works for HK/Macau/Taiwan, China. But I am not sure if those will solve the region problem, do you know?