MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #232: No internet connection error, even i have internet connection.

jwtiyar, 2024-12-19:
Hey
Iam facing this issue and i don't know what is the problem every time tells me check internet connection:
`
[2024-12-19] [05:02:12] [INFO] - ************************************
[2024-12-19] [05:02:12] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-12-19] [05:02:12] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-12-19] [05:02:12] [INFO] - ************************************
[2024-12-19] [05:02:12] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-12-19] [05:02:12] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-12-19] [05:02:12] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-12-19] [05:02:12] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-12-19] [05:02:12] [INFO] - My Blog: https://www.neko.ink/
[2024-12-19] [05:02:12] [INFO] - ************************************
[2024-12-19] [05:02:12] [INFO] - Starting ADB server...
[2024-12-19] [05:02:13] [INFO] - Processing device 75ff9747(1)...
[2024-12-19] [05:02:13] [INFO] - Finding BootLoader unlock bind request...
[2024-12-19] [05:02:14] [INFO] * Now you can bind account in the developer options.
[2024-12-19] [05:02:29] [INFO] - Account bind request found! Let's block it.
[2024-12-19] [05:02:29] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in C:\php\bypass.php on line 333
[2024-12-19] [05:02:29] [INFO] - Sending POST request...
[2024-12-19] [05:02:29] [INFO] ! Fail to send request, check your internet connection.
PS C:\php>

valerian2, 2024-12-24:
même problème de mon côté avec le redmi note 13 pro 5g
[image]

kuromoci, 2025-01-06:
same problem here

Jaxx9527, 2025-03-16:
dont use the hotspot form the phone you want to unlock BL