MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #233: HP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt()

nck404, 2024-12-22:
Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\Admin\Pictures\unblock\Bypass-1.0-fix-universal\bypass.php:230
Stack trace:
#0 C:\Users\Admin\Pictures\unblock\Bypass-1.0-fix-universal\bypass.php(330): decryptData('#&^sdFi8B3veIi6...')
#1 {main}
  thrown in C:\Users\Admin\Pictures\unblock\Bypass-1.0-fix-universal\bypass.php on line 230

Sypers, 2024-12-27:
This is thrown because the PHP can't find the openssl library. see where PHP is expecting to find the ext files, try placing the PHP files in "`C:\php`" since that where it was searching and it worked for me.

vishalpj14, 2024-12-27:
> This is thrown because the PHP can't find the openssl library. see where PHP is expecting to find the ext files, try placing the PHP files in "`C:\php`" since that where it was searching and it worked for me.

Works perfectly

ginolaginola, 2024-12-27:
> > Ezt azért dobja ki, mert a PHP nem találja az openssl könyvtárat. nézze meg, hogy a PHP hol várja az ext fájlokat, próbálja meg elhelyezni a PHP fájlokat a " `C:\php`" mappába, mivel ott keresett, és nekem működött.
> 
> Tökéletesen működik

For Poco X6 Pro not work:(
What files copy and where?

Sypers, 2024-12-28:
> For Poco X6 Pro not work:(
> What files copy and where?
> [image]

This is a different issue. It indicates that the device has to go through the account verification process. This unfortunately can't be fixed.

ginolaginola, 2024-12-29:
> > > > Ezt azért dobja ki, mert a PHP nem találja az openssl könyvtárat. nézze meg, hogy a PHP hol várja az ext fájlokat, próbálja meg elhelyezni a PHP fájlokat a " `C:\php`mappába, mivel ott keresett, és nekem működött.
> > > 
> > > 
> > > Tökéletesen működik
> > 
> > 
> > A Poco X6 Pro esetében nem működik:( 
> > Milyen fájlokat másolnak és hova?
> > ![Névtelen](https://private-user-images.githubusercontent.com/114621984/399009195-fb873014-14ed-4ec1-afc3-8abd204da72f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1MDE3MjAsIm5iZiI6MTczNTUwMTQyMCwicGF0aCI6Ii8xMTQ2MjE5ODQvMzk5MDA5MTk1LWZiODczMDE0LTE0ZWQtNGVjMS1hZmMzLThhYmQyMDRkYTcyZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyOVQxOTQzNDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YjRjZGQ2MWM5ZmJkNGMxMjQ1MmQwYTBiMzBkMzdmNjhlN2VjMjJkY2YxY2FlOTU2MjczNjgxZjMwNWJkNTcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.AjN4jF3ZDno8B_VJhw5bG80hi0qIAaMy6qxGFTmnIkM)
> 
> Ez más kérdés. Azt jelzi, hogy az eszköznek át kell mennie a fiókellenőrzési folyamaton. Ez sajnos nem javítható.

Many thanks. The unlock will work from xiaomi community app?

Sypers, 2024-12-29:
> Many thanks. The unlock will work from xiaomi community app?

Yes, but it's very hard to apply to unlock the bootloader.