MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #242: Cant unlock, decryptData() fails

volteret4, 2025-01-08:
[2025-01-08] [14:59:27] [INFO] - Finding BootLoader unlock bind request...
[2025-01-08] [14:59:27] [INFO] * Now you can bind account in the developer options.
adb: invalid transport id
[2025-01-08] [14:59:27] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /home/huan/tt/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 330 and defined in /home/huan/tt/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php:228
Stack trace:
#0 /home/huan/tt/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php(330): decryptData()
#1 {main}
  thrown in /home/huan/tt/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 228

dabiceaqui, 2025-01-08:
YOYOYO I MADE A VIDEO FOR DA FIX FR

https://youtu.be/8nu1VoCK5Ew

mdan96, 2025-01-11:
@dabiceaqui thank you.

dabiceaqui, 2025-01-11:
> @dabiceaqui thank you.

np, thought i could help

VitmaSense, 2025-09-16:
@icey0814  Couldn't fix error 230 ((