MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #245: Xiaomi 14 Hyper OS 2.04.0

sali9043, 2025-01-19:
I have followed the instructions but I got error when I click on Add Account script get error.
[2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO] [2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO] - [2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:00] [INFO]

Π

X

- [2025-01-19] [18:38:00] [INFO] - Starting ADB server...

[2025-01-19] [18:38:00] [INFO]

[2025-01-19] [18:38:01] [2025-01-19] [18:38:01] [INFO]

* Xiaomi HyperOS BootLoader Bypass *

* By NekoYuzu

Version 1.0 *

GitHub: https://github.com/MlgmXyysd

XDA: https://xdaforums.com/m/mlgmxyysd.8430637

X (Twitter): https://x.com/realMlgmXyysd PayPal: https://paypal.me/MlgmXyysd

My Blog: https://www.neko.ink/

- Processing device 4c1edbf7(2)...

[INFO] Finding BootLoader unlock bind request...

Now you can bind account in the developer options.

[2025-01-19] [18:38:11] [INFO] Account bind request found! Let's block it.

[2025-01-19] [18:38:11] [INFO] - Refactoring parameters...

PHP Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\sali9\Downloads\php\bypass.ph ✔p: 230

Stack trace:

#0 C:\Users\sali9\Downloads\php\bypass.php(330): decryptData('#&^qTSCuLmS4D5Y...')

#1 {main}

thrown in C:\Users\sali9\Downloads\php\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\sali9\Downloads\php\bypass.php:230 Stack trace: #0 C:\Users\sali9\Downloads\php\bypass.php (330): decryptData('#&^qTSCuLmS4D5Y...') #1 {main}

thrown in C:\Users\sali9\Downloads\php\bypass.php on line 230

C:\Users\sali9\Downloads\php>

sali9043, 2025-01-19:
i was able to solve this after moving this to C:\php but now i am getting this error 
C:\php>bypass.cmd
[2025-01-19] [19:27:35] [INFO] - ************************************
[2025-01-19] [19:27:35] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2025-01-19] [19:27:35] [INFO] - * By NekoYuzu          Version 1.0 *
[2025-01-19] [19:27:35] [INFO] - ************************************
[2025-01-19] [19:27:35] [INFO] - GitHub: https://github.com/MlgmXyysd
[2025-01-19] [19:27:35] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2025-01-19] [19:27:35] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2025-01-19] [19:27:35] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2025-01-19] [19:27:35] [INFO] - My Blog: https://www.neko.ink/
[2025-01-19] [19:27:35] [INFO] - ************************************
[2025-01-19] [19:27:35] [INFO] - Starting ADB server...
[2025-01-19] [19:27:36] [INFO] - Processing device 4c1edbf7(1)...
[2025-01-19] [19:27:36] [INFO] - Finding BootLoader unlock bind request...
[2025-01-19] [19:27:37] [INFO] * Now you can bind account in the developer options.
[2025-01-19] [19:27:43] [INFO] - Account bind request found! Let's block it.
[2025-01-19] [19:27:43] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333
[2025-01-19] [19:27:43] [INFO] - Sending POST request...
[2025-01-19] [19:27:44] [INFO] - Request parameter error (10000)

[image]

ManikSharma69420, 2025-01-20:
I HAVE FIXED BYPASS CODE:

<?php
/**
 *
 *    Copyright (C) 2002-2024 NekoYuzu (MlgmXyysd) All Rights Reserved.
 *    Copyright (C) 2013-2024 MeowCat Studio All Rights Reserved.
 *    Copyright (C) 2020-2024 Meow Mobile All Rights Reserved.
 *
 */

/**
 *
 * Xiaomi HyperOS BootLoader Bypass
 *
 * https://github.com/MlgmXyysd/Xiaomi-BootLoader-Bypass
 *
 * Bypass Xiaomi HyperOS community restrictions of BootLodaer unlock account bind.
 *
 * Environment requirement:
 *   - PHP 8.0+
 *   - OpenSSL Extension
 *   - Curl Extension
 *   - ADB
 *
 * @author MlgmXyysd
 * @version 1.0
 *
 * All copyright in the software is not allowed to be deleted
 * or changed without permission.
 *
 */

/***********************
 *    Configs Start    *
 ***********************/

// Global flag
// If you are running a Global ROM (Non-China Mainland), set it to true
$useGlobal = false;

/*********************
 *    Configs End    *
 *********************/

/***************************************
 *               WARNING               *
 *    Do NOT modify the codes below    *
 *               WARNING               *
 ***************************************/

// Include php-adb library
// https://github.com/MlgmXyysd/php-adb

require_once __DIR__ . DIRECTORY_SEPARATOR . "adb.php";

use MeowMobile\ADB;

/*************************
 *    Constants Start    *
 *************************/

global $api;
global $sign_key;
global $data_pass;
global $data_iv;

$api = $useGlobal ? "https://unlock.update.intl.miui.com/v1/" : "https://unlock.update.miui.com/v1/";
$sign_key = "10f29ff413c89c8de02349cb3eb9a5f510f29ff413c89c8de02349cb3eb9a5f5";
$data_pass = "20nr1aobv2xi8ax4";
[... 299 lines cut ...]
    case 0:
        logf("Target account: " . $res["data"]["userId"], "g");
        logf("Successfully blocked bootloader unlock binding request.", "g");
        break;
    case -3:
        logf("Request has been canceled.", "y", "!");
        break;
    case -100:
        logf("This device is not supported.", "r", "!");
        break;
    case -105:
        logf("You are running Global ROM (non-CH), BootLoader is permanently locked.", "r", "!");
        break;
    default:
        logf("Unknown response code: " . $res["code"], "r", "!");
        break;
}

logf("Done.");

BUT NOW THIS:

[image]

SAD

sali9043, 2025-01-20:
i was able to fix unknown reponse code :10000 after enabling 

extension_dir = "ext" extention php.ini file now it now it stuck at this like  
* Now you can bind account in the developer options.

[image]

[image]

ManikSharma69420, 2025-01-20:
I think its because of our settings app

https://github.com/TheAirBlow/HyperSploit

try this and you'll see an error

[image]

so yeah..

also downgrading doesnt work

[image]

[image]

SAD :((((

ManikSharma69420, 2025-01-20:
BUT I do have a different HyperOS version

[image]

havent upgraded yet 😉 

and im not planning to 😉

Marcade80, 2025-01-21:
@ManikSharma69420 I did not upgrade either; (also have Xiaomi 14, HyperOS 1.0); but Hypersploit eventually also gave me "Request parameter error (10000)".

Did you get around it yet?

sali9043, 2025-01-21:
> [@ManikSharma69420](https://github.com/ManikSharma69420) I did not upgrade either; (also have Xiaomi 14, HyperOS 1.0); but Hypersploit eventually also gave me "Request parameter error (10000)".
> 
> Did you get around it yet?

You problem is easy to solve as you can install settings.apk software which has the bug which let you Bypass so you can download it from release section of this repo but mine is hyper OS 2 which doesn't support that setting apk because it was made for Android 14

liaofuyu1, 2025-01-24:
Did you finally unlock the xiaomi 14 bootloader? Is it the China Version? Appreciate if you tell me

sali9043, 2025-01-25:
No bro I didn't unlock and china mainland OS they have stopped bootloader
unlock for them.

Now doing offical way increasing my level to 5 and then apply for
bootloader unlock china time 00:00

ManikSharma69420, 2025-01-27:
> You problem is easy to solve as you can install settings.apk software which has the bug which let you Bypass so you can download it from release section of this repo but mine is hyper OS 2 which [...]

Bro I can't install the app if you CAN please share the apk with me, many thanks
Because I have seen many settings.apk but it doesn't work on my phone (sad)

sali9043, 2025-01-27:
Try see if this work or not mentioned
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip

sali9043, 2025-01-27:
> I experience the same on HyperOS 2.0.3.0. I don't think it's the error in
> script but rather the way that (bad) Xiaomi blocked bootloader unlock. The
> error message appears when account is about to be binded and on a device
> there's a message that account and device are from different regions
> (meaning - CN device and non-CN account = you cannot unlock the device).
> However I'm still searching other options... By the way - I tried to
> install Settings app but a downgrade is not possible (even when I
> uninstalled original app via adb command, an old Setting app for Android 14
> cannot be installed - it is automatically blocked).

That setting app ment for hyper OS 1.0 not for us hyper OS 2.0

martinezix00, 2025-01-28:
> That setting app ment for hyper OS 1.0 not for us hyper OS 2.0
> […](#)

Yeap, but the thread name is "[Xiaomi 14 Hyper OS 2.0.4.0 Request parameter error (10000)" so I decided to share my observation as well ;)

sali9043, 2025-01-29:
Okay

ManikSharma69420, 2025-01-29:
> Try see if this work or not mentioned
> https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip
> […](#)

it doesn't.
that was actually the settings app i was trying to install when it said "it cant be installed"

sali9043, 2025-01-29:
Yes it for hyper os 1.0 android version 14

sugarskiee, 2025-02-03:
is there any solution guys?

sali9043, 2025-02-03:
Not that I know of.

sugarskiee, 2025-02-03:
> Not that I know of.
> […](#)

im so hopeless with this hyper 2.0

sali9043, 2025-02-03:
> Not that I know of.
> … <#m_-1021352464284549182_>
>
> im so hopeless with this hyper 2.0

Me too! I sold my xiaomi 14 for the same price I bought due to this.

sugarskiee, 2025-02-03:
> Me too! I sold my xiaomi 14 for the same price I bought due to this.
> […](#)

now u use ?

sali9043, 2025-02-03:
OnePlus 8 Pro

sugarskiee, 2025-02-03:
> OnePlus 8 Pro
> […](#)

okayy, i think im gonna wait for a while, maybe xiaomi would open his eyes 🥲

sali9043, 2025-02-03:
Just do level up to 5 and your xiaomi comunity app account try applying
chinese 12:00 AM time daily you might be lucky too unlock

sugarskiee, 2025-02-03:
> Just do level up to 5 and your xiaomi comunity app account try applying
> chinese 12:00 AM time daily you might be lucky too unlock
> […](#)

i think its imposible for hyper 2.0

ManikSharma69420, 2025-02-06:
any luck, anyone?

sugarskiee, 2025-02-06:
nah bro still stuck on this unlocking shit

sugarskiee, 2025-02-06:
> > [image]
> > YEAHHHHHHH FINALLY GUYS
> 
> So ... what did you do to make it work?

i did nothing bro, just randomly add account on mi unlock status, bcs i've already have the permission on mi community

martinezix00, 2025-02-06:
> > > YEAHHHHHHH FINALLY GUYS
> > 
> > 
> > So ... what did you do to make it work?
> 
> i did nothing bro, just randomly add account on mi unlock status, bcs i've already have the permission on mi community

Hi and congratulations :)
Please confirm one thing - do you have chinese device (purchased device dedicated for chinese market) and you live outside China? Or do you have a global device and live outside China?
I've checked with ncunlock and with halabtech and they said that CN version of Pad 6S with HyperOS 2.x is not supported even with EDL mode (I tested EDL myself and I can confirm that EDL cable in my case does not help)

sugarskiee, 2025-02-06:
> Hi and congratulations :) Please confirm one thing - do you have chinese device (purchased device dedicated for chinese market) and you live outside China? Or do you have a global device and live [...]

mine is global version bro, i feel bad if u have chinesse version maybe u cant unlock it forever even with paid service 🥹 :), im so happy rn cant wait 72 hours to unlock

sugarskiee, 2025-02-06:
the only way is sell ur chinesse phone and buy one the global version dude

metty99432, 2025-02-06:
[image] Has anyone solved it, I'm in the same situation

martinezix00, 2025-02-06:
Has anyone solved it, I'm in the same situation

You can't do anything. CN device outside China cannot be unlocked. At least for now. Maybe there will be an exploit in the future...

ManikSharma69420, 2025-02-09:
Bruhh
Imma try soon

blackshark555666, 2025-02-10:
[image]
my internet connection was good
idk, why this error appears

blackshark555666, 2025-02-10:
the reason why i used xiaomi phone over10 years ago, because of this bootloader is easy to unlock. but now, since they changed the rules, everything looks like terrible.

mueasier, 2025-02-19:
I tried all the methods I found and came to a conclusion:
1. Due to the change in the encryption method of the setting application, you need to install the old version of setting to bypass
2. The old version of setting is built on A14. If you have upgraded to A15, you cannot downgrade and install it
3. This problem first appeared in January 2024, when it was still A14, but now OS2 has been upgraded to A15, so it can be considered that there is no useful setting application available
It's time to change your phone

martinezix00, 2025-02-19:
> I tried all the methods I found and came to a conclusion:
> 1. Due to the change in the encryption method of the setting application, you need to install the old version of setting to bypass
> 2. The [...]

Hi
I've tested it. Even if you're on Android 14, switching Settings app will not help. I tested it on CN and on global xiaomi device. Xiaomi did a "great" job (meaning, that there's no way to bypass this unfortunatelly) with locking the bootloader on newer devices, especially, those with HyperOS. We can only wait for a new exploit that may do the job, but you never know if that will happen. It seems that what we can do, is to show them the red card by not purchasing their devices...

martinezix00, 2025-02-21:
> As far as I understand, in the new “Settings” application, information about the device is encrypted with other keys. If find these keys, you can adapt the method for new phones models 

That's how I understand it. But even old app does not help in most cases. I tried this on cn and global pad6S. I'm afraid that hyperOS brought more bad changes than one could imagine. It was much easier to "crack" miui. By the way, finding keys is rather cryptographic problem and it is not an easy task to "find a key"

yurec0098, 2025-02-21:
> That's how I understand it. But even old app does not help in most cases. I tried this on cn and global pad6S. I'm afraid that hyperOS brought more bad changes than one could imagine. It was much [...]

The current method uses encryption keys, they were taken from somewhere ...

jarwonozt, 2025-03-26:
> [image] Has anyone solved it, I'm in the same situation

It seems like the bypass has been fixed by Xiaomi

selltool, 2025-04-01:
[image]
Xiaomi fixed

selltool, 2025-04-01:
[image]
Fixed

RedmiT4Pro, 2025-07-21:
> [image] Fixed

How did you manage to have it fixed?

DarthiDu, 2026-05-24:
@mylanezu , did you find a solution? Have the same problem