MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #247: Hey can u help me out about parameter error 10000 in redmi note 13 pro 5G. I am stuck and scrolling through forums for 2 days tried almost every method

connorvb14, 2025-01-23:
I have tried adding that thing before L330, and got that error. i cant install any older versions of settings but i managed to install one settings version. even though adb sideload showed success and i cant see any diffrence in the settings app and the error keeps coming.

this was error code before adding **var_dump($args);**

[2025-01-23] [16:09:23] [INFO] - ************************************
[2025-01-23] [16:09:23] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2025-01-23] [16:09:23] [INFO] - * By NekoYuzu          Version 1.0 *
[2025-01-23] [16:09:23] [INFO] - ************************************
[2025-01-23] [16:09:23] [INFO] - GitHub: https://github.com/MlgmXyysd
[2025-01-23] [16:09:23] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2025-01-23] [16:09:23] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2025-01-23] [16:09:23] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2025-01-23] [16:09:23] [INFO] - My Blog: https://www.neko.ink/
[2025-01-23] [16:09:23] [INFO] - ************************************
[2025-01-23] [16:09:23] [INFO] - Starting ADB server...
[2025-01-23] [16:09:24] [INFO] - Processing device d8cc2a12(1)...
[2025-01-23] [16:09:24] [INFO] - Finding BootLoader unlock bind request...
[2025-01-23] [16:09:24] [INFO] * Now you can bind account in the developer options.
[2025-01-23] [16:09:41] [INFO] - Account bind request found! Let's block it.
[2025-01-23] [16:09:41] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 332

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 332
[2025-01-23] [16:09:41] [INFO] - Sending POST request...
[2025-01-23] [16:09:42] [INFO] - Request parameter error (10000)

and this was the error after i added **var_dump($args);**

[2025-01-23] [16:15:13] [INFO] - ************************************
[2025-01-23] [16:15:13] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2025-01-23] [16:15:13] [INFO] - * By NekoYuzu          Version 1.0 *
[2025-01-23] [16:15:13] [INFO] - ************************************
[2025-01-23] [16:15:13] [INFO] - GitHub: https://github.com/MlgmXyysd
[2025-01-23] [16:15:13] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2025-01-23] [16:15:13] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2025-01-23] [16:15:13] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2025-01-23] [16:15:13] [INFO] - My Blog: https://www.neko.ink/
[2025-01-23] [16:15:13] [INFO] - ************************************
[2025-01-23] [16:15:13] [INFO] - Starting ADB server...
[2025-01-23] [16:15:14] [INFO] - Processing device d8cc2a12(1)...
[2025-01-23] [16:15:14] [INFO] - Finding BootLoader unlock bind request...
[2025-01-23] [16:15:14] [INFO] * Now you can bind account in the developer options.
[2025-01-23] [16:15:17] [INFO] - Account bind request found! Let's block it.
[2025-01-23] [16:15:17] [INFO] - Refactoring parameters...
string(1292) "#&^mSVryD80p3bHDJhDROJJFn4lHkPe9dFuMCAasuLQv7WpQeLlPnUeIF+WqULTuBlOWgKbWKgRaT2NnlZ6hTSruZcSj1PGvIdDsNJROSxoyMWFZobp5Io8DXMjDLjp0SDr2TJhrkCovCXcU4JaZt+kclL+7P+Umivtk2mPafwjRCOrsFkre3K/PT8ikCqpcTagkHmU61QcH73dnPqofHw51RBtJqGRf8WYca8fYvv2Zf6erzrY45zvuufTf6UmWldzC3UTw7s43sPBgx762TvtOw0hiGQMX8spQKdmOdh/VuBPvKWg96SxmTEanm/Uzpk/ebSUI+wWdkBGdBS8P3Cavg==!!AwFjhWu7rwh6+72CVONT6pPfvuuAkSnVH9ICzp47XJInmUD0yiPtisDkE4a3mMVFTiGR5VOSGCNGsSVCN8IcDNib8h1SFld33fY5BSEDWAQ0rleod0ZYOe+rKs2r3c4vCasOOvksoCGIASY9tIaNBFL/v4Is2P4K/TeWk0SdtvmhytEbRvZxxzhbo6tSa+LDytiQiPcnSFFG05gTW15KeXOwGtRewu0W77RxnknEz/0r9aks6RGZH1E/1LzfTkbkzF8a6W6DebXjlFfjjwMfzju2Rk6wutkpeNzvJbl55/xRyCI5wG8jQ6m2RP7tswNMw0Dw4Dek/iejrLJNqfjRE5TZvTFfTYKTs5o9hwwUKgE5LrorrL0v2AP7PyKgvtLfnFMFQLsMFGaf3d+7pNx2LWPaXdjZe/kSr5eyLD4PkXz0k8Gg4kf5zPUXAibNzI9lu0qwk/Pw+vEIUcAnz737oI5M9Q+QooE/p7cyqvlblbuQRB3TxKn72I+cgletVXDZiyVNOup4lH+yks/RF47wGjKkYRqJejT/33JefBNn3DcOcKiDy+8BC3gqmICBbPgdG4kMUF9iW3ZRLTcBg1P6kQoFwxFT0wBQC1efEO2NcvE/Xd8HewBHC0+hgSVQQa6k/K3Bvo0N8Su7Qj0k7s0y+dlEV+B2Ey02yPqRdifdmObydj2CtIRHQsTzZTot3yqRGaAQgE0lIuTloOXAQWu7y0so4hpk1DfwbhOnTPyc1RD1c9AqQHqkXAowtYSF1/E7RWU+mK8Jf0AnEaEKF77M9gcbpIsW7+Uzr7bw175hSobz9WtwBHtl5nys+2ar5Sx6bXdvc1Cfcnxi/hpo/j8e5ADdOmLN7/A07X8etCYmmUBHuTPzjKYvJWeD2PNznIxSzUiQIhMhV4kLsPgKlbB6q5ggCtkbvYFR0kyHfUZJaRA=^&#"
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 332

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 332
[2025-01-23] [16:15:17] [INFO] - Sending POST request...
[2025-01-23] [16:15:18] [INFO] - Request parameter error (10000)

I done everything and tried everything. can u please help me out

Joseisg, 2025-01-25:
same problem here... theres no setting.apk for redmi 13 5g either

firejackal, 2025-01-26:
Same issue here ... the request parameter is not the issue, it's the prior line "Trying to access array offset on value of type null "

Looking over the code and doing some debugging, there is encrypted data going into the decryptData() function, but there nothing is coming out (it's being assigned null.)
That line in decryptData() has me assuming maybe xiaomi changed the encryption for the latest revisions of hyperos.

sali9043, 2025-01-29:
Goto php.ini file enable extension= "ext"

connorvb14, 2025-01-29:
> Goto php.ini file enable extension= "ext"

cant see a extension named ext

[image]

connorvb14, 2025-01-29:
> Same issue here ... the request parameter is not the issue, it's the prior line "Trying to access array offset on value of type null "
>
> Looking over the code and doing some debugging, there is [...]

I Saw a thread with a tool to retrieve encryption key, if xiomi changed the key can we use that tool to find the key and mod the bypass.php?? i am below noob level in these kind of things just asking

mamiiblt, 2025-02-03:
i bought an garnet 2 weeks ago too, but almost every method doesnt work anymore...