MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #253: Can't install settings.apk to fix error 10001 on HyperOS 1.0.16.0UNRCNXM

Samiiya, 2025-02-25:
Anyone know how to fix this issue? I have been trying to install so many settings.apk but all are failed to install.

I also tried to install on different file manager app but can't install on unofficial app too.

I'm using Redmi Note 13 Pro 5G with china ROM HyperOS 1.0.16.0UNRCNXM

2aq112x35, 2025-02-26:
maybe version is older than phone`s.😢

[image]

Samiiya, 2025-02-27:
> maybe version is older than phone`s.😢
> 
> [image]

Wish someone could make a version 35 or higher so we can crack the settings to unlock bootloader 😢