MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #260: POCO C65 Request parameter error (10000)

arsenyno, 2025-03-13:
[2025-03-13] [17:34:37] [INFO] - ************************************
[2025-03-13] [17:34:37] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2025-03-13] [17:34:37] [INFO] - * By NekoYuzu          Version 1.0 *
[2025-03-13] [17:34:37] [INFO] - ************************************
[2025-03-13] [17:34:37] [INFO] - GitHub: https://github.com/MlgmXyysd
[2025-03-13] [17:34:37] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2025-03-13] [17:34:37] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2025-03-13] [17:34:37] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2025-03-13] [17:34:37] [INFO] - My Blog: https://www.neko.ink/
[2025-03-13] [17:34:37] [INFO] - ************************************
[2025-03-13] [17:34:37] [INFO] - Starting ADB server...
[2025-03-13] [17:34:38] [INFO] - Processing device (maybe important thing)
[2025-03-13] [17:34:38] [INFO] - Finding BootLoader unlock bind request...
[2025-03-13] [17:34:38] [INFO] * Now you can bind account in the developer options.
[2025-03-13] [17:34:42] [INFO] - Account bind request found! Let's block it.
[2025-03-13] [17:34:42] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in C:\Users\Арсений\Desktop\HyperOS-BootLoader-Bypass-win64\bypass.php on line 333

Warning: Trying to access array offset on null in C:\Users\Арсений\Desktop\HyperOS-BootLoader-Bypass-win64\bypass.php on line 333
PHP Deprecated:  str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in C:\Users\Арсений\Desktop\HyperOS-BootLoader-Bypass-win64\bypass.php on line 333

Deprecated: str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in C:\Users\Арсений\Desktop\HyperOS-BootLoader-Bypass-win64\bypass.php on line 333
[2025-03-13] [17:34:42] [INFO] - Sending POST request...
[2025-03-13] [17:34:43] [INFO] - Request parameter error (10000)

Please help, version of HyperOS is 1.0.7.0 (UGPRUXM)

hsutungc, 2025-03-16:
same issue

Jaxx9527, 2025-03-16:
try to install the settings apk
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/tag/1.0-fix

arsenyno, 2025-03-16:
> try to install the settings apk https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/tag/1.0-fix

Now when open Developer Settings app crash

arsenyno, 2025-03-16:
Idk how it working

jarwonozt, 2025-03-22:
same.. seems like you have to queue for UBL permit quota at Mi Community

arsenyno, 2025-03-22:
Idk where need to tap there

> same.. seems like you have to queue for UBL permit quota at Mi Community
I've been trying to do this for over a month now, but without success.

arsenyno, 2025-03-22:
Idk how to reply and don't close it

> same.. seems like you have to queue for UBL permit quota at Mi Community

I've been trying to do this for over a month now, but without success.

closed 2025-03-22 by arsenyno