MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #267: Problema de región.

jhonpauljhons1, 2025-04-24:
Tengo un dispositivo Xiaomi 13 ultra (Ishtar) con rom modificada global. Pero es dispositivo chino. Este no me deja de loquear, dice que el dispositivo está fuera de la región donde se registró, este mensaje salta cuando agrego cuenta y dispositivo corriendo el script. Ya intenté crearme una cuenta con región de china y también usando vpn tanto en dispositivo como en la PC. Alguna idea, necesito ayuda.

Qynklee, 2025-05-04:
I got some advices from other person. If you had Chinese Xiaomi account, you need revert ROM back to China ROM (maybe using EDL mode). After that, you can unlock bootloader with this account.

jhonpauljhons1, 2025-07-30:
Cómo puedo acceder a una cuenta con permiso EDL?