MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #273: Request parameter error (10000)

emir1st1, 2025-05-09:
On Redmi Pad I get Request parameter error (10000) downgrading settings app is impossible I tried everything is there a way to fix this?

emir1st1, 2025-05-10:
> 可能是解密算法和密钥不符合当前系统的版本

How do I fix it then?

NeverTeaser, 2025-05-11:
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/78/files reinstall `setting`

emir1st1, 2025-05-11:
> https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/78/files reinstall `setting`

Coolapk link doesn't work and I tried to install the apk provided in GitHub it doesn't install. It says settings isn't updatable.

NeverTeaser, 2025-05-11:
> > https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/78/files reinstall `setting`
> 
> Coolapk link doesn't work and I tried to install the apk provided in GitHub it doesn't install. It says settings isn't updatable.

I used adb tool install

emir1st1, 2025-05-11:
> > > https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/pull/78/files reinstall `setting`
> > 
> > Coolapk link doesn't work and I tried to install the apk provided in GitHub it doesn't install. It says settings isn't updatable.
> 
> I used adb tool install

I use adb too as I said it says not updatable

Andreunto, 2025-05-22:
> I use adb too as I said it says not updatable

Hi, u solve this problem? I'm in same situation

emir1st1, 2025-06-11:
> > > Can you try disabling miui optimization in developer settings? It may make settings upgradable.
> > 
> > 
> > I tried it too still no luck
> 
> Oh. Seems that Xiaomi has fixed that. In the past I have seen someone downgraded settings successfully after disabling it.

So there is no way to downgrade settings app right?

4682neo, 2025-06-11:
虽然跟这个项目没关系，我想说是有一种方法，那就是9008，一些旧的设备，使用这种方式最快，无论你是什么版本！当然风险还是有的！我最后也这个办法解决的

4682neo, 2025-06-11:
你可以自己去了解一下，他是芯片底层的刷，我也是自己找的材料，相关的信息网上都有

emir1st1, 2025-06-11:
> 你可以自己去了解一下，他是芯片底层的刷，我也是自己找的材料，相关的信息网上都有

What even is 9008?

4682neo, 2025-06-11:
我不是技术大佬，所以没法提供更多信息，这个具有一定风险，激活9008需要特定方法

4682neo, 2025-06-11:
> > 当然联发科也有类似方法
> 
> My device has MediaTek Helio G99? What can I do?我的设备有 MediaTek Helio G99？我该怎么办？
我也是这个，联发科的

4682neo, 2025-06-11:
MTK  sp_flash _tools

4682neo, 2025-06-11:
> MTK sp_flash _tools

这是联发科的

haha077, 2025-08-04:
fix it now？