MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #279: How to downgrade the installation settings? What method do you use? In order to solve the request parameter error 10000, I don't understand anything

WANG6892, 2025-07-02:
I learned from the Internet that the error code 10000 needs to install the old version of the settings. I have obtained the installation package of the old version of the settings, but I have exhausted all the solutions, but I still can not downgrade the installation. I have not seen the Xposed module that can be used. Please guide me how to downgrade the installation settings application？

carrieyz, 2025-07-20:
> The older Settings.apk can be only installed on Android 14, if you're using HyperOS2 (based on Android 15) ,you can do nothing.较旧的 Settings.apk 只能安装在 Android 14 上，如果您使用的是 HyperOS2（基于 Android 15），则什么也做不了。

我的是红米note11Tpro，OS是HyperOS2.0.5.0，但是Android版本显示是14，Settings.apk还是无法安装，试过很多APK了都不行。

carrieyz, 2025-07-20:
> The older Settings.apk can be only installed on Android 14, if you're using HyperOS2 (based on Android 15) ,you can do nothing.较旧的 Settings.apk 只能安装在 Android 14 上，如果您使用的是 HyperOS2（基于 Android 15），则什么也做不了。

sorry，my english is bad。translate：
My phone is Red Rice Note11Tpro, OS is HyperOS 2.0.5.0, but Android version shows 14, Settings.apk still can't be installed, tried a lot of APK, but it doesn't work.

xitebanned-byte, 2025-10-18:
> In later versions of HyperOS 1 and onwards Xiaomi used different encryption algorithms that made such tools can't forge requests to the server. If you're still on HyperOS 1, use this link [here](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip) to patch Settings app (install via ADB) and proceed on computer as normal. If you're on HyperOS 2 or later, or downgrading Settings does not work, you're out of luck if you're on a Chinese device because there's not a Settings apk for Android 15 that uses the old encryption algorithm. [image]
> 
> 在之后版本的 HyperOS 1 及更高版本中，小米使用了不同的加密算法，使得此类工具无法伪造对服务器的请求。如果您仍在使用 HyperOS 1，请使用此处的[链接](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip)修补“设置”应用（通过 ADB 安装），然后在电脑上正常操作。如果您使用的是 HyperOS 2 或更高版本，或者降级"设置"后，并且使用的是中国设备，则暂无方法，因为安卓15版本的设置apk就没有使用旧加密方法的。
> 
> Edit: Some said use 9008 mode for SnapDragon or MTK sp_flash _tools for MediaTek, but I don't endorse it because it's much harder and involves more risks to your devices and some newer devices have blocked these methods. 有人说骁龙处理器使用 9008 模式，联发科使用 MTK sp_flash _tools，但我不赞成这样做，因为它更难，并且会给你的设备带来更多风险，而且一些较新的设备已经阻止了这些方法。

There was a problem while parsing pakage.(Redmi note 11 pro+ 5g // hyper os 1) Helppp

Royce33, 2025-10-29:
> In later versions of HyperOS 1 and onwards Xiaomi used different encryption algorithms that made such tools can't forge requests to the server. If you're still on HyperOS 1, use this link [...]

谢谢你的回复，我最后的解决办法是刷机到MIUI，然后申请解锁成功的。关于HyperOS2我尝试了很多办法，的确没法直接破解，只能通过刷机解决。